JonnyHaystack/HayBox #1: Inability to compile with fewer buttons

Ddhuet, 2021-01-23:
I'm attempting to use this firmware on my Smashbox, which lacks a select and home button. If I set these to NULL in the pinout file, it throws errors. If I delete the lines they throw errors.

JonnyHaystack, 2021-01-23:
Can you show the errors? Also that reminds me I should add a smashbox pinout file..

JonnyHaystack, 2021-01-25:
When you say "delete the lines" do you just mean in the pinout file? You have to delete them from the pinout file *and* from HayB0XX.ino (and your setup file too I guess)

Ddhuet, 2021-01-25:
I was just deleting them from the pinout file, or trying to NULL them. I will try it again soon with deleting from the other sources. It would be nice if there was a convenient way to mark them as not needed. I don't want random pins activating, and I would imagine most DIY boxes don't have home/select buttons.

JonnyHaystack, 2021-01-25:
Agreed, I would much prefer this to work without any manual modifications. I will have to think about the best way to do that

ribbanya, 2021-07-13:
> I will have to think about the best way to do that

Wouldn't a `#define` in each of the `*_setup.h` files, and a `#if` around the references to `SELECT` and `HOME`, make the most sense? Then the user would only have to uncomment one line somewhere, or it could be predefined in certain header files (like `setup_smashbox.h`).

JonnyHaystack, 2021-07-13:
> 
> 
> > I will have to think about the best way to do that
> 
> Wouldn't a `#define` in each of the `*setup_*.h` files, and a `#ifdef` around the references to `SELECT` and `HOME`, make the most sense? Then the user would only have to uncomment one line somewhere, or it could be predefined in certain header files (like `setup_smashbox.h`).

That would then make it specific to only adding or removing those two buttons, which is not what I want. It's better to just set the unused buttons to an unused pin if possible.