JonnyHaystack/HayBox #4: Getting compilation errors when targeting Smash Box

ribbanya, 2021-07-13:
## Errors

```log
In file included from X:\Cpp\HayB0XX\HayB0XX\DInputBackend.cpp:1:0:
X:\Cpp\HayB0XX\HayB0XX\src\ArduinoJoystickLibrary\src\Joystick.h:32:2: error: #error The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).
 #error The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).
  ^~~~~
In file included from X:\Cpp\HayB0XX\HayB0XX\DInputBackend.cpp:4:0:
DInputBackend.h:16:3: error: 'Joystick_' does not name a type
   Joystick_ *mpJoystick;
   ^~~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\DInputBackend.cpp: In constructor 'DInputBackend::DInputBackend()':
DInputBackend.cpp:8:3: error: 'mpJoystick' was not declared in this scope
   mpJoystick = new Joystick_(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_GAMEPAD,
   ^~~~~~~~~~
DInputBackend.cpp:8:20: error: expected type-specifier before 'Joystick_'
   mpJoystick = new Joystick_(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_GAMEPAD,
                    ^~~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\DInputBackend.cpp: In destructor 'virtual DInputBackend::~DInputBackend()':
DInputBackend.cpp:26:3: error: 'mpJoystick' was not declared in this scope
   mpJoystick->end();
   ^~~~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp: In constructor 'KeyboardMode::KeyboardMode(socd::SocdType, state::InputState&)':
KeyboardMode.cpp:9:3: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.begin();
   ^~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp:9:3: note: suggested alternative: 'KeyboardMode'
   Keyboard.begin();
   ^~~~~~~~
   KeyboardMode
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp: In destructor 'virtual KeyboardMode::~KeyboardMode()':
KeyboardMode.cpp:13:5: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
     Keyboard.releaseAll();
     ^~~~~~~~
DInputBackend.cpp:27:10: error: type '<type error>' argument given to 'delete', expected pointer
   delete mpJoystick;
          ^~~~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\DInputBackend.cpp: In member function 'virtual void DInputBackend::SendOutputs(state::OutputState)':
DInputBackend.cpp:32:3: error: 'mpJoystick' was not declared in this scope
   mpJoystick->setButton(0, outputState.b);
   ^~~~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp:13:5: note: suggested alternative: 'KeyboardMode'
     Keyboard.releaseAll();
     ^~~~~~~~
     KeyboardMode
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp: In member function 'void KeyboardMode::Press(char, bool)':
KeyboardMode.cpp:24:9: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
         Keyboard.press(key);
         ^~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp:24:9: note: suggested alternative: 'KeyboardMode'
         Keyboard.press(key);
         ^~~~~~~~
         KeyboardMode
KeyboardMode.cpp:26:9: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
         Keyboard.release(key);
         ^~~~~~~~
X:\Cpp\HayB0XX\HayB0XX\KeyboardMode.cpp:26:9: note: suggested alternative: 'KeyboardMode'
         Keyboard.release(key);
         ^~~~~~~~
         KeyboardMode
exit status 1
'Joystick_' does not name a type
```

## [Differences](https://github.com/JonnyHaystack/HayB0XX/compare/76202489bfa5745beecb8b111fd0451be498d13d...cdcee7be0be4ed590d7f0285a6f8c5036e4eeaca) from `master` (76202489bfa5745beecb8b111fd0451be498d13d)

```diff
diff --git a/HayB0XX/HayB0XX.ino b/HayB0XX/HayB0XX.ino
index d5be178..2e03ae1 100644
--- a/HayB0XX/HayB0XX.ino
+++ b/HayB0XX/HayB0XX.ino
@@ -10,7 +10,7 @@
     Zac Staples' Arduino_Vector library
 */
 
-#include "setup_gccpcb2.h"
+#include "setup_smashbox.h"
 
 #include "DefaultKeyboardMode.h"
 #include "FgcMode.h"
@@ -75,8 +75,8 @@ void readInputs() {
   gInputState.mod_x = (digitalRead(pinout::MODX) == LOW);
   gInputState.mod_y = (digitalRead(pinout::MODY) == LOW);
   gInputState.start = (digitalRead(pinout::START) == LOW);
-  gInputState.select = (digitalRead(pinout::SELECT) == LOW);
-  gInputState.home = (digitalRead(pinout::HOME) == LOW);
+  // gInputState.select = (digitalRead(pinout::SELECT) == LOW);
+  // gInputState.home = (digitalRead(pinout::HOME) == LOW);
   gInputState.b = (digitalRead(pinout::B) == LOW);
   gInputState.x = (digitalRead(pinout::X) == LOW);
   gInputState.z = (digitalRead(pinout::Z) == LOW);
@@ -114,8 +114,8 @@ void setup() {
   pinMode(pinout::MODX, INPUT_PULLUP);
   pinMode(pinout::MODY, INPUT_PULLUP);
   pinMode(pinout::START, INPUT_PULLUP);
-  pinMode(pinout::SELECT, INPUT_PULLUP);
-  pinMode(pinout::HOME, INPUT_PULLUP);
+  // pinMode(pinout::SELECT, INPUT_PULLUP);
+  // pinMode(pinout::HOME, INPUT_PULLUP);
   pinMode(pinout::B, INPUT_PULLUP);
   pinMode(pinout::X, INPUT_PULLUP);
   pinMode(pinout::Z, INPUT_PULLUP);
```

## Arduino settings

[image]

## Other notes

I'm on Windows 10 and using the `git clone --recursive` method.

JonnyHaystack, 2021-07-13:
From the README:

[image]

This also applies to smashbox if it uses an ATMega2560 because that MCU does not support native USB

closed 2021-07-13 by JonnyHaystack

ribbanya, 2021-07-14:
Ah, I missed that. Thank you.

JonnyHaystack, 2021-07-14:
Well, it wasn't written as a step for smash box until now. I just added it in because you brought it to my attention, so thanks for that :smile:

ribbanya, 2021-07-14:
Is there a way to utilize the Gen 2's USB joystick capabilities? Is the way Smash Box does it significantly different from how the B0XX does it?

JonnyHaystack, 2021-07-14:
Umm I don't really know much about smashbox hardware tbh. Do the different generations have different MCUs? Maybe newer ones use the 32U4?

ribbanya, 2021-07-14:
No, it's still the Arduino Mega 2560, but it has a custom bootloader and a separate processor for USB mode.

https://cdn.shopify.com/s/files/1/0166/4408/files/Smash_Box_Experience_Notes_Mar.23.2020.pdf

https://www.hitboxarcade.com/products/r3-smash-box-upgrade-board

I asked in Discord but they seem unwilling to explain how it works on the firmware end.

> Gen2 boards have an extra processor specifically for USB Host functions.
> 
> It's an Arduino Mega with custom firmware. If you got a gen2 board in your smash box gen2, the gen2 upgrade or a smash box experience kit, you have a board with USB game pad capabilities.
> 
> I can't do much if someone else's custom firmware lacks those features.

JonnyHaystack, 2021-07-19:
Hmm I don't really know enough about the hardware and I don't own a SmashBox so it would be hard for me to add support for this. Maybe if you can find out more about the USB host board and whether it can be driven using some open source library, then it could be possible. I think it would require new communication backends as well (which isn't hard, if you have a library doing the heavy lifting).