JonnyHaystack/HayBox #19: HID issues on Linux, macOS

JonnyHaystack, 2022-08-06:
On Linux, some of the coordinates (mostly higher magnitude ones it seems) are off by 1. E.g. cardinals are 0.9875 (79) instead of 1.0 (80)

On macOS, there have been reports from users that it isn't showing up as a HID gamepad at all, but keyboard mode is working fine.

Probably going to have to compare against other HID report descriptors again (e.g. Arduino HID) to figure out a solution. Linux issue possibly fixable through Dolphin profile change though?

Widdershin, 2022-08-19:
Happy to help investigate and test issues on macOS (and linux if needed), will be running HayBox on my new LBX when it arrives next week.

JonnyHaystack, 2022-08-19:
Thanks! I'm fine testing on Linux, but macOS testing would be really appreciated

Widdershin, 2022-08-19:
Good to know. I’ll follow up once my LBX arrives next week!

JonnyHaystack, 2022-09-26:
The Linux issue is fixed on master branch, and I think macOS issue might've been fixed at some point but hard to say with all the different macOS versions behaving differently. I'll close this for now until I hear otherwise.

closed 2022-09-26 by JonnyHaystack