JonnyHaystack/HayBox #21: Spams buttons when swapping to a Keyboard mode on a Pico

UMS-Ultra, 2022-09-03:
When inputting the button combination for a Keyboard mode on a Pico, the device will rapidly spam those same 3 keys for maybe a second. For example, when inputing the Default Keyboard mode, it will spit out A, F, and H repeatedly in a long sequence. The sequence is variable, and appears to be related to how long all 3 buttons where held down _after_ the button combination was inputed.

JonnyHaystack, 2022-09-10:
This is known, it is caused by the keyboard mode being rapidly constructed/destructed many times very quickly

ribbanya, 2023-01-21:
Would something like this fix the issue? I wrote this for Smash Box rocker toggles but it also ensures that you can't switch to the current mode.

https://github.com/JonnyHaystack/HayBox/blob/3459f03b3f41987e806ef094073a03cef9dd3e52/config/mode_selection.hpp#L70-L98

JonnyHaystack, 2023-01-25:
Yeah, it makes it significantly more complicated for the average user to edit the mode bindings though, which I'd prefer to avoid. I feel like the real problem is just the way keyboard modes themselves exist in the current architecture. It doesn't really fit in well. Especially considering they don't even work with XInputBackend which is the default on Pico now, so you need to hold a button on plugin as well. Gonna have to redesign it at some point.

commit 66ed1bf2 (2023-10-21): feat: track current game mode config index

Track index of current game mode config to prevent switching from one
GameModeConfig to the same GameModeConfig, which in turn prevents
spamming of keypresses when entering a keyboard mode on RP2040.

Closes #21
closed 2025-03-15 by JonnyHaystack