JonnyHaystack/HayBox #24: [Discussion]: Possible improvements, code and stick value questions

avahe-kellenberger, 2022-10-10:
I have been editing code for the Ultimate mode to change/fix a few things:

### Queuing dash into ftilt using modY + cstick results in a dash attack
This means the `leftStickX` value is too high. By reducing the multiplier by 1, it started working properly.

But, this piqued my interest in the values, the differences between HayBox and the stock B0XX firmware, and how these values work. It seems that directly assigning a value to `outputs.leftStickX` does not work as I would expect; the value ends up being different at runtime (NOTE: I am testing these values using SDL). **I'm curious why this is, and if there's a way to set a value I'd expect directly (e.g. `9680`)**

The default values for the B0XX firmware for left/right while holding modY are `-9681` and `9680` respectively, while HayBox's values are `-13435` and `13434`. 

From B0XX's Smash Ultimate manual, `"Modifier Y contains the slowest
walk in the game."` I'm not sure if this is actually true, as with some testing I've been able to walk with lower values, although I'm not sure if the walking speed was actually slower. This does beg the question though; with HayBox's mod Y, are we walking as slow as the game will allow? Or, is this even a goal of using mod Y with HayBox?

Also, `leftStickX` values without any modifiers are much higher, akin to an actual gcc.

---

I'm more than happy to test out differences between the firmware, an official gcc that I own, etc. if you'd like more data.

JonnyHaystack, 2023-01-19:
Think I might've responded to this on Discord I forget, but yeah the goal was never to use the exact same values as the B0XX or for Mod Y to give the slowest walk. I didn't create the Ult mode myself anyway, because I didn't have access to the game (and even now I don't really play it ever). Also if you were testing with the Switch USB backend, there's something wrong with the scaling on that currently. Not sure exactly why it doesn't map the values 1:1, and the scaling doesn't seem to be linear. If you would be able to figure out what's going on with that it would definitely be helpful. The main difficulty is not knowing of a way to display the stick values that the Switch/Ult is actually ending up with (either through GC adapter or direct USB).

ribbanya, 2023-01-19:
Wouldn't it be not that hard for GitHub Actions to upload the binaries whenever master is updated?

JonnyHaystack, 2023-01-19:
@ribbanya I think you replied to the wrong issue by accident maybe? And yeah, I just also wanted it to ideally attach versioned binaries to a release whenever I create one.

avahe-kellenberger, 2023-01-25:
> The main difficulty is not knowing of a way to display the stick values that the Switch/Ult is actually ending up with (either through GC adapter or direct USB).

I have a program that logs these values from a controller, if that'd be useful for you I could set that up.
Then maybe you could set each button to be an incrementing value of the left stick's x axis for instance, and try to determine the scale?

For this issue `queuing dash into ftilt using modY + cstick results in a dash attack` I think that is resolved by #31