JonnyHaystack/HayBox #35: Please add location for slippi controller .ini files for mac to the ReadMe

SchillerQuinn, 2023-01-31:
Please add instructions for the .ini files for Mac into the ReadMe

The location is 

`/Users/USER/Library/Application\ Support/Slippi\ Launcher/netplay/Slippi\ Dolphin.app/Contents/Resources/Sys/Config/Profiles/GCPadSlippi\ Dolphin.app/Contents/Resources/Sys/Config/Profiles/GCPad`

To get into to it, you have to first go to `/Users/USER/Library/Application\ Support/Slippi\ Launcher/netplay/Slippi\ Dolphin.app/Contents/Resources/Sys/Config/Profiles/GCPad`, right click on Dolphin.app, and click "open contents", then you have to navigate to `Contents/Resources/Sys/Config/Profiles/GCPad` and put the .ini files there.

JonnyHaystack, 2023-02-01:
Thanks, I'm confused about the "open contents" thing though. Is it not just an ordinary file path?

closed 2023-05-21 by JonnyHaystack

Soundtoxin, 2023-07-12:
I'd like to suggest one edit to what went into the README for this. You put a generic <USER>, but if we use the terminal you can actually use the variable `$USER` and it will supply the current logged in user there when running the command, so people could just copy paste it. In addition I would replace the "Cmd + Shift + G" part with using the `open` terminal command.
So instead it could say something like this:

"On Mac: Open Terminal and run `open "/Users/$USER/Library/Application Support/Slippi Launcher/netplay/Slippi Dolphin.app/Contents/Resources/Sys/Config/Profiles/GCPad"`"

(quotes needed around the path due to the spaces)

That will open it in Finder the same way but it can be copy-pasted into the terminal so it's a bit easier potentially. The CMD-Shift-G method requires manually typing out the path so you can select your user, or going back after pasting to fix it. Some people might not even know what their username is since it's not displayed many places.

JonnyHaystack, 2023-07-17:
Thanks for the suggestion! Sounds like a good idea.

commit a1684a8e (2023-05-21): Add Mac Slippi controller profile path to README

Resolves #35