JonnyHaystack/HayBox #36: PS3/PS4/PS5 Support

Soundtoxin, 2023-01-31:
Are there any plans to add PlayStation support? Would that be possible? I'm not sure if I can have one controller that does it all or if I need at least 2 with different firmware on each.

JonnyHaystack, 2023-02-01:
PS3 is partially tested (with DInput backend) and the inputs do at least work although I don't know if they're mapped correctly. Native PS4 and PS5 support is never going to happen because they use encryption that it's maybe illegal to bypass (very lame, I know). You can just buy a Brook Wingman XE adapter though. That works on PS3, PS4, and with PS4 games played on PS5. Someone has tested HayBox with a Wingman and it worked with the Switch backend at least.

For full PS5 support you'd need to dual mod with a Brook UFB + the PS5 compatibility expansion board I believe, which is significantly more complicated.

Soundtoxin, 2023-06-30:
I have a PS3 now and can confirm that the DInput mode works on it, although the mappings for DInput FGC Mode are not correct. Tested with Skullgirls' Training Mode as well as the "GamePad Test" homebrew app.

[image]

Here's a picture of a Mayflash arcade stick that has both the PlayStation and Xbox labels in case that helps with bindings.

I think Cross is bound correctly (bottom left) but the other three face buttons (square, triangle, circle) are all in the wrong spots. Circle is where triangle should be, square is where circle should be, triangle is where square should be.
R1 is where R2 should be, R2 is where L2 should be,  L1 is where R1 should be, L2 is where L1 should be.

So the two rows should look like this:
|| | | |
| :---: | :---: | :---: | :---: |
| Square| Triangle | R1 | L1 |
| Cross | Circle | R2 | L2 |

Currently they are:
| | | | |
| :---: | :---: | :---: | :---: |
| Triangle | Circle | L1 | L2 |  
| Cross | Square | R1 | R2 |

Start is start. The directional bindings seem fine, left/down/right in the usual spots, up on Mod-X but also C-Up, makes sense so you can hit up with either thumb.

C-Left seems to be bound to select, which is kinda useful since that's normally not bound at all in other modes. C-Right, C-Down, what would normally be A in Smash, and Mod-Y don't seem to be bound to anything from what I can tell. It would be useful to have the PS button bound to something if there's extra room somewhere.

I also suspect there are some issues in the other modes for DInput, like the face buttons being in the wrong spots again. In Rivals mode, the top left most button that is usually L in Smash appears to be R2, I can't find anything bound to L2 in Rivals Mode on DInput. Z button appears to be L1. The Light/Medium shield buttons are bound to R3 and L3 instead of being different amounts of pressure on the triggers. Right stick doesn't seem to work and is being detected like I'm holding the right stick up, at least in the PS3 GamePad Test program.
Switching to Melee Mode, the L button activates both R2 at full push and moves the right stick to the right, or maybe up and right. Similar weirdness with the R button activating R1 at full pressure and moving the right stick down (relative to its resting position, which again might be pushed up by default). LS/MS buttons move the right stick down different amounts.
In real usage on a PS3 you might not use the Melee Mode, you'd definitely want the FGC mode, and probably either Rivals/Ultimate if you wanted to try to play a normal game that uses both sticks (at least I find those mods tend to work better universally than Melee Mode which I think doesn't push the stick as far and can cause issues).

I'd like to re-open and re-title this issue, hopefully that's okay.

I have been using my Flatbox rev 5 with GP2040-CE firmware to play my PS3 a lot lately (Soul Calibur IV, Skullgirls) and it would be great if I could also use the Frame1 with it since the spaced out buttons are more comfortable. I'm guessing DInput and PS3 are not used too commonly by Frame1 users, so these issues were probably not noticed by many people.

JonnyHaystack, 2023-07-04:
Thanks for all the testing/info. I'll look into making a new comms backend with the inputs remapped to work on PS3. There's no set standard for how HID gamepad inputs should be mapped, so it can't be a "one size fits all" solution.

Soundtoxin, 2023-07-05:
>There's no set standard for how HID gamepad inputs should be mapped, so it can't be a "one size fits all" solution.

Interesting. I didn't realize that. Is there any other device that needs DInput testing then?

Soundtoxin, 2023-09-12:
You may or may not know this, but after the last release, my inputs no longer do anything on a PS3 in DInput mode. I assume this is related to the default DInput device changing to the Leonardo. The inputs were wrong before, so not like I was getting use out of it on a PS3 anyway, just thought I'd re-test this issue after updating and report back.

JonnyHaystack, 2023-09-12:
That was only a change to the Dolphin profile, so it wouldn't affect usage on PS3. Not sure what else could've changed.