JonnyHaystack/HayBox #40: two mod_x buttons?

Tree52, 2023-02-10:
Want two mod_x buttons but only one of them works :( Maybe there's a way and I'm just dumb.

[image]

AtomToast, 2023-02-14:
There can only be one of each button in your layout. I also had to ask Haystack for help on this one. 
So the only way is to create more modifiers. Take a look at this commit from my personal fork to see how you could possibly do this https://github.com/AtomToast/HayBox/commit/b06eebd24c311646525b4ed497f828d2a9dcc4fd

Tree52, 2023-02-14:
Thanks :D I got it working. Making bool mod_x_pressed is smart, rather than just doing mod_x || mod_x1 everywhere like I probably would have done, haha. Closing the issue.

closed 2023-02-14 by Tree52

JonnyHaystack, 2023-03-04:
> Thanks :D I got it working. Making bool mod_x_pressed is smart, rather than just doing mod_x || mod_x1 everywhere like I probably would have done, haha. Closing the issue.

You could also just do `inputs.mod_x = inputs.mod_x || inputs.mod_x1` if you want to keep changes minimal and you want them to be treated exactly the same.