JonnyHaystack/HayBox #42: Is it possible to send a button hold on boot on GameCube?

Soundtoxin, 2023-02-19:
I have a GameCube running Picoboot and I've configured it so that if you hold B on a controller when turning it on, it will go straight to Game Boy Interface instead of Swiss. I can't seem to get this to work with my Open-Frame1 running HayBox. Also, even when Swiss loads, I find that I can't move through the menu until I unplug the cable and plug it back in. Does the GameCube have to already be on and do you then have to plug in the controller for it to work normally?
In my case the controller is already plugged in on both ends, I hold the B button, then I press the GameCube's power button.
Am I perhaps just running into a conflict? I see in the readme that holding B on plugin goes into Brook board mode. So if the controller isn't receiving power until the console turns on, maybe that's when it counts as a plug-in, and then because I'm holding B it goes into that mode.
It seems like I also can't move through the menus if I boot while not holding anything until I replug the cable. Do I have a wonky cable/port or is this expected behavior? Once it's working, it keeps working fine, but I think I have been running into these sorts of initialization issues now and then.

JonnyHaystack, 2023-03-26:
The autodetection relies on receiving a poll shortly after startup. So if it's receiving power before the console is sending polls commands, it will end up falling back to XInput mode.
It will also ignore the first poll, which might affect your Picoboot button hold on startup thing. Hard to say without looking at the code for that. You could change how the autodetection works in `joybus_utils.cpp` if you want to simplify it, or you could just do away with autodetection and make it so you have to hold a button on plugin to choose the comms backend.

Brook board mode isn't a thing in the default Pico config so that's not the problem.

Soundtoxin, 2023-06-30:
Thanks for the explanation. Going to close this issue for now.

closed 2023-06-30 by Soundtoxin