JonnyHaystack/HayBox #44: Default "DefaultKeyboardMode" not working with LBX

Animach1ne, 2023-03-04:
Hello,
I am using a LBX plugged to a Windows 10 PC via USB (so Dinput should be default). I can switch between input modes just fine, and I managed to remap buttons successfully (thank you!)
However the Keyboard mode that comes with the latest build doesn't seem to fulfil its purpose - no keys actuate.
The latest build seem to have assigned the keyboard key "A" to "L", "B" to "Left" etc... But maybe some extra modifications are needed to get it to work. I'm sorry I can't figure it out by myself (lack of programmation skills), the documentation does say that the <UpdateKeys()> is where things need to be changed, but it's unclear to me what needs to be done.
Thanks for your help

JonnyHaystack, 2023-03-09:
What button combination are you pressing to enter keyboard mode? It's still working fine for me on my HTangl V1.1 (which is an ATMega32U4 based controller), so I see no reason why there would be any issues on LBX. I would also be surprised if it was an issue and the guys at JFCA hadn't brought it up despite having a support contract in place with me...

Animach1ne, 2023-03-12:
Thanks for your answer :)
According to LBX's manual it's Mod Y + Start + L. When doing that, I indeed switch modes - no gamepad inputs are detected (even though the LBX is still detected as a gamepad, by gamepadviewer for instance).
I can switch back to another controller mode with no issue. It's just, nothing happens in keyboard mode.
I have just noticed another possibly interesting thing : if I unplug the LBX while in keyboard mode, it's not working anymore upon replugging, whatever the controller mode. I have to re-flash the firmware to get it to work.

JonnyHaystack, 2023-03-13:
Have you tried on a different computer? This sounds more like an issue with the OS and not with the controller. You can't brick the firmware by going into keyboard mode. Modes are not persistent.

Animach1ne, 2023-03-26:
It was indeed an OS issue : I have tried on a laptop with Windows 7 and it worked.
So I digged into common keyboard issues with Windows 10, and unchecked a power option which was likely causing the issue. It's all good now !
Thanks a lot for your help :)

JonnyHaystack, 2023-03-26:
Glad to hear you got it working, and thanks for sharing your solution

closed 2023-03-26 by JonnyHaystack

Animach1ne, 2023-06-08:
> Any chance you could provide more details on the solution? similar spot; not sure what option you are referring too!
> 
> Thanks

I think it was in the device manager. If you right click on your device, you should have power supply options available on the rightmost tab. Try checking/unchecking boxes there!