JonnyHaystack/HayBox #45: Create a macro to enable training mode restart

AtomToast, 2023-03-09:
Currently I don't think it is possible to restart training mode, since the hardcoded shortcut for that is L+R+A. At least when using Ultimate mode via usb.
Since there is no actual L button on most controllers. So currently I have created this macro for myself (https://github.com/AtomToast/HayBox/commit/f3000f2070e59602cc846a84d3365487070211f3), where I can press mx+my+z to also trigger buttonL and a.

Do you think something like this would make sense in the main firmware too?
Are there any legality concerns?
Alternative implementation/buttons?

JonnyHaystack, 2023-03-09:
I've been informed about this problem by a few users at this point, and was thinking in the next release I'd just bind LS and MS to L and R so you could press LS + MS + A to reset.

AtomToast, 2023-03-09:
I guess that would be a good solution with the default layout. Though technically you'd only need an L button, since z already maps to R

commit 5e7eea11 (2023-05-21): Add LS and MS = L and R in Ultimate mode

Resolves #45
closed 2023-05-21 by JonnyHaystack