JonnyHaystack/HayBox #46: Windows not recognizing LBX / Startup Help

GaffeyTaffy, 2023-03-23:
Hi, I am buying a friends LBX and they had someone else configure it for them. This LBX currently has crane's flashed onto it and I am trying to use it on my Windows 11 PC. I figured a good first step would be to upgrade to HayBox. I don't have much experience with VSC. I used GIT to clone the repository and followed the steps on the main page. After searching and trying different things I still had no change and would get the following error. 
[image]
I went to device manager and changed the COM port being used to make sure it wasn't being used elsewhere. I am stumped. The LBX shows up as Arduino Leonardo in devices, but will not show up as a gamepad. Any help would be much appreciated.

JonnyHaystack, 2023-03-26:
It doesn't look like you selected a serial port on the bottom bar (see where it says "Auto" in your screenshot). Sometimes it doesn't detect correctly, so you could try selecting it manually.

By the way, this is now the official firmware that ships with new LBXs so you should be going to JFCA for support with using it on their products. They should be able to provide pre-built binaries that you can flash without having to build from source. If they don't know about how to do that I'll have to talk to them.

closed 2023-03-26 by JonnyHaystack

GaffeyTaffy, 2023-03-26:
That didn't work either, but I will contact them! Thank you.