JonnyHaystack/HayBox #47: Default FGC mode has likely-unintended mixed L/R and LT/RT layout

cynidea, 2023-04-27:
So the standard 8-button layout found in something like Street Fighter is:
<pre>
X  Y  *R*   _L_
A  B  *RT*  _LT_
</pre>

However, some folks (myself included) prefer having the \*R\*s and _L_s swapped like so:
<pre>
X  Y  _L_   *R*
A  B  _LT_  *RT*
</pre>
Though I personally prefer the latter, I understand that the former is industry standard.

Here's the issue: neither of these layouts are implemented in the current release. Instead, we have:
<pre>
X  Y  _L_   *R*
A  B  *RT*  _LT_
</pre>
...which I presume is unintended.

JonnyHaystack, 2023-04-27:
Sounds like you've swapped your midshield and lightshield pins, because it's correct in the code. Lightshield is on ring finger, midshield is on pinky.

cynidea, 2023-04-27:
Woah wait really? That's completely my bad then, I was operating off of BubbleBox layout and just presumed stock Haybox used the same.

closed 2023-04-27 by cynidea