JonnyHaystack/HayBox #54: Mode Memory (on unplug)

Soundtoxin, 2023-06-30:
The GP2040-CE firmware remembers your last mode when you unplug the controller, which I find to be pretty handy. I personally use Melee Mode less than the other modes, so if I could have it remember Ultimate/Rivals mode, it would save me a lot of trouble. This would probably matter even more in a tournament setting where you may be re-plugging the controller a lot as you change who's playing at a setup and when. For people who primarily play Melee, it would just remember they last used the Melee Mode and should work basically the same for them, so I don't think there's much to lose here.

Soundtoxin, 2023-07-01:
I just had the thought that this could potentially conflict with the automatic mode picking in some cases, so maybe the memory would only apply to scenarios where detection-based switching didn't come into play.

JonnyHaystack, 2023-07-04:
That is one problem that it would introduce, among others. Sometimes people accidentally get into the wrong mode and it becomes harder to troubleshoot things when you add that possibility, and you can't tell what the user may have persisted in flash/EEPROM. I'd rather just introduce a configurator program in future that can change default mode tbh, but that's a long way off.

Soundtoxin, 2023-07-04:
I did almost open an issue about adding a web configurator (would make the keyboard mode a lot more useful), but wasn't sure how hard that would be.

I do still like the idea of a mode memory for situations where I'm playing the same game a few hours and possibly reconnecting the controller. It could be a PS3 game one day and a PC game another, so not something I'd likely wanna tweak my default mode for each time. I understand if it's too difficult to implement with how things are set up right now.

JonnyHaystack, 2024-02-13:
For info, HayBox 3.0 will have support for configurator apps via [HayBox-proto](https://github.com/JonnyHaystack/HayBox-proto), as well as being able to change the default mode, comms backend, and SOCD via an OLED display.

Of course, most controllers don't have an OLED display but it's a feature I implemented for the Glyph. It's an intuitive way to make either temporary or persistent config changes. Other people will be able to use the configurator. You will be able to create new on-plugin binds for different gamemodes as well, if you only use a couple of gamemodes and don't like using the on-the-fly button combos.

closed 2025-04-10 by JonnyHaystack