JonnyHaystack/HayBox #55: Release schedule

Soundtoxin, 2023-06-30:
I'm curious if there's a rough plan for when the next release will be, and how often you intend to make future releases. I understand there are some big changes in the works like correcting values in Ultimate Mode that you may want to wait on before releasing. Changes like the controller identifier that was used changing to fix the dpad issue on Linux would be great to have in a release. I unfortunately don't find building with PlatformIO to be very accessible, so I feel kind of stuck on the last stable release for now. (I should probably try to figure out building HayBox someday, as I think that's the only way to add a new mode, and it would be cool to make a new mode for rhythm games like Taiko no Tatsujin.)

JonnyHaystack, 2023-07-04:
The main things holding back this release are the XInput vendor/product ID change which needs a new release of the B0XX input viewer in order for it to not break input viewer compatibility, and as you say the Switch backend scaling, which has proved to be quite a difficult problem. That one is pretty close at this point though, and it's better than it was in the last release, so it wouldn't really be a blocker for the next release.

Aside from that it's just been a really busy year for me and I've not been working on this as actively as I would like. Got plenty of new feature ideas I want to implement in the near future.

Soundtoxin, 2023-07-04:
The VID/PID change and fixing the DInput mappings (mentioned this in another issue about PS3 support but was unable to reopen it) would make my controller way more useful right away (Would make FGC Mode usable on both my Steam Deck and PS3 then), so would be amazing to get all that stuff in the next release. Looking forward to the future of HayBox!

I guess this issue could be closed when the next release happens, or now since you did answer my question, either way.

closed 2023-09-12 by JonnyHaystack