JonnyHaystack/HayBox #60: Sonic Superstars (Switch) - can't select Chaos Emerald ability

Soundtoxin, 2023-10-19:
They want you to use the right stick flicked in one of 8 directions in a radial menu to select an ability. For some reason with my Open-Frame1 running HayBox it won't select the ability using the c-stick buttons. I can change where the arrow points, but the menu stays stuck open, similar to if you keep the stick held on a regular controller. I figured maybe it was a stick angle issue, but I don't notice any different behavior when using the Mod X/Y buttons. I was using Ultimate mode, but I don't notice a major difference with the other platform fighter modes. Maybe it expects the stick to return to center slightly more gradually instead of jumping back to no stick input. Using a more traditional controller (8bitdo Ultimate controller) the menu closes after I flick the stick in a direction and it then selects that ability.
Could c-stick behavior be altered so that it emulates the stick returning to center with stick values the whole way as if you'd quickly flicked it? I imagine if it worked that way, it would still be able to work the same in Smash as well.
This game is *almost* completely playable even in FGC mode except for this weird right stick requirement, it's a real shame I can't find a way to make it work. From what I've seen the PC version lets you bind buttons to changing the abilities left/right around the menus, but I couldn't find a way to configure the controls on the Switch version, so I think you're forced to just use the defaults, including right stick for that ability menu.

Soundtoxin, 2023-10-19:
Possibly related to #41 
>Also, because the stick cannot fully progress from Neutral to Up, scrolling up is not possible in certain Switch System Settings.

Sounds like a similar issue if I'm understanding it right.

JonnyHaystack, 2024-02-13:
This will hopefully be fixed by the improvements to analog stick value scaling for Switch backend, which will be in the next release. If you could try the `fix/switch_backend_scaling` branch and provide any feedback that would be appreciated.

Soundtoxin, 2024-02-15:
I tested the branch and it seems to have the same issue with the menu being stuck open. Any right stick button pressed leaves the menu stuck open instead of it closing after release. I noticed also that pressing R, which cycles through the stuff in the wheel has the same issue. The menu never disappears, so you can't actually select anything. The only way to make the menu go away after a single press of R or the C buttons seems to be to pause and unpause the game.

JonnyHaystack, 2025-04-10:
@Soundtoxin Are you able to reproduce this on 3.0.0 or is it fixed now? The Switch backend fixes have been included in that release.

Soundtoxin, 2025-04-11:
Can confirm I still get the same behavior, one press of any c-stick direction keeps the radial menu stuck open and it never selects the power. Same thing in both Ultimate and Rivals 2 mode. Plugged in with USB directly.

Testing again with the GC adapter, surprisingly it actually *does* work as expected, selecting the Emerald Power and closing the menu as soon as I release the button. I didn't even know I could use that outside of Smash. Changing between Melee/Ultimate/Rivals 2 mode using the GC adapter (if that even works) doesn't seem to affect the behavior either.