JonnyHaystack/HayBox #71: XInput not working with pico (but all other backends do)

glicnak, 2024-06-08:
I am able to get the controller to register on my pc by holding Z as a DInput Controller. It also works on my switch by holding X plugged into the dock, and as a GCC controller by plugging into the adapter. 

However, if I simply plug the usb into my computer without holding anything (which should use XInput backend), nothing happens. The USB device is unidentified ("Unknown USB Device (Invalid Configuration Descriptor)"). I have tried reinstalling my drivers, and I have also tried it on different PC's, all giving the exact same error... I am kind of at a loss of what to do next.. Any help would be greatly appreciated!

JonnyHaystack, 2024-06-08:
What did you flash your controller with exactly? Sounds like you've downloaded the latest release zip and built that, or are building from outdated master branch?

glicnak, 2024-06-08:
I used the latest release zip

JonnyHaystack, 2024-06-08:
Yeah old release zips from 2.4.0 and prior don't work because the arduino-pico version was not pinned, and the Adafruit TinyUSB Arduino library annoyingly seems to have been updated in such a way that it broke the XInput functionality, so pinning the arduino-pico version is necessary for now. Use the master branch and it should work fine. It'll have to redownload all the dependencies again though when you open it in VS code so wait for it to do that first.

glicnak, 2024-06-08:
I will try that! Btw while I have you here:

1: Thank you so much for this project and all your help!

and 2: I tried to add new buttons, and it seems to be working. I just want to know if I did it properly.. What I did was: 
Include-core-state -> add buttons
Src-comms-b0xxinputviewer -> add buttons and change report [#] to reflect the new amount of Scan Inputs
Include-comms0b0xxinputviewer -> change report [#] to match above
Config-pico-config.cpp → remap buttons to pins

Were the b0xxinputviewer changes useless?

JonnyHaystack, 2024-06-08:
1 - no problem 😄

2 - you can't add buttons to the input viewer report because the B0XX input viewer protocol only supports 20 buttons. Not sure if that's what you were trying to do but if you just wanted extra buttons to map and don't care about the input viewer, you don't need to change anything in there at all.

3. Yeah there's some weird scaling the Switch does to GameCube adapters which needs to be replicated for the Switch USB backend because it doesn't scale most USB gamepads in the same way. Getting it exactly right is tricky, especially with rounding error being a factor. See #31 for discussion/info on this topic.

glicnak, 2024-06-08:
Master branch worked :)

Oh okay that makes sense. So if I want to add buttons and I don't care about the viewer, all I need is: 
Include-core-state -> add buttons
Config-pico-config.cpp → map buttons to pins
And that's it? Awesome!

I'll check it out and see what I can do! I'm replicating eSam's smashbox mod, and right now it works perfectly on gamecube mode so that's good enough for me.

Thanks so much dude!!

JonnyHaystack, 2024-06-08:
Yeah pretty much. Also, you may want to check out the configurator branch (which is the current development branch), as it supports a lot more buttons by default and has a powerful new configuration system.

closed 2024-06-08 by JonnyHaystack