JonnyHaystack/HayBox #73: Configurator Branch - Inputs for mode change / default mode

glicnak, 2024-07-15:
I've been trying to make my own modes using the configurator branch, but I can't seem to find where to change the default modes when plugging into switch / pc... and I also can't find where to change the button inputs to change modes on the fly... There is no "mode_selection.hpp" in config... and the one in core (both src and include) doesn't seem to have this.. Any help would be appreciated!!

glicnak, 2024-07-15:
Note: I am able to make a new mode and have it compile... I just don't know where to set the inputs to change to it.. also I'd want it to be the default for Xinput config, but I can't find that either :\

JonnyHaystack, 2024-07-29:
You can do this through the configurator now. Hold c-down on plugin for configurator mode and see https://github.com/JonnyHaystack/haybox-cli and https://haybox.dev/proto for documentation on config format

closed 2025-03-20 by JonnyHaystack