JonnyHaystack/HayBox #74: Gamecube adapter not working with latest version of haybox device

garciaErick, 2024-07-31:
Expected behavior:
Connect haybox device to a valid gc adapter and be able to play on PC Slippi with this adapter.

Actual behavior:
Box device only works via usb mode, on gamecube adapter it does not work

Troubleshooting steps:
* Updated to latest version of haybox firmware
* Confirmed device is usable (played tekken, melee with usb connection)
* Bought mayflash adapter (its V07)

Before you go on and explain why I should use USB connection, please focus on the conversation. I have my use case and my reasons, I want to ensure this feature does work and want to root cause why it is not working for me.

garciaErick, 2024-07-31:
Attempted to try to install firmware 05 to mayflash since somebody mentioned that works however I havent gotten past this prompt

JonnyHaystack, 2024-08-26:
You've given no information about what hardware you're running HayBox on

garciaErick, 2024-08-26:
It is an LBX from junkfood arcades.

JonnyHaystack, 2024-08-26:
Then you should be going to them for support first. But the first things I'd suggest to check are that both of the USB connectors on the mayflash adapter are plugged in, and that you're plugging in while holding the A button.

garciaErick, 2024-08-26:
It was the same issue with my actual b0xx device as well, can you confirm this works on your end?

JonnyHaystack, 2024-08-26:
Yes. Just to clarify, was it ever working for you?

JonnyHaystack, 2024-08-26:
Well this has always worked in HayBox, so I can only conclude it's an issue with your setup

garciaErick, 2024-08-27:
Hmm it appears the issue is that I need v05 of the mayflash adapter but I am unable to update. Is there any reason why we forcibly need to use v05 of the mayflash? Can't this be fixed in the software? I bought a new Mayflash adapter only to be blocked again by the 07 firmware and no actionable steps to downgrade. 

Got any advice to get an adapter that works? I will be trying to return mine I guess :/

garciaErick, 2024-08-27:
Do you have a list of gamecube adapter and versions that might work?

garciaErick, 2024-08-27:
Will be trying tomorrow with the official Nintendo adapter: https://www.amazon.com/Nintendo-GameCube-Controller-Adapter-Switch/dp/B07DQWN8Y8)

garciaErick, 2024-08-31:
Works with the official super nintendo adapter. As an action item for this "resolved" issue could we add the list of current supported devices and versions in the README?

garciaErick, 2024-08-31:
Thanks for the support

closed 2025-04-10 by JonnyHaystack