JonnyHaystack/HayBox #78: SOCD woes - changed defaults?

Soundtoxin, 2024-08-15:
Did the SOCD cleaning behavior change in the configurator branch compared to master? I was playing some Rivals and kept missing moves/inputs and dying, eventually realized if I'm holding left, then hold right, then let go of right, it does not go back to holding left, it does nothing instead. It's like it goes from "last wins" to "do nothing". I compared to a GP2040-CE device and that works how I expect. I hadn't used my HayBox device much lately but I feel like this behavior is new. I don't know if you play Rivals but mainly it resulted in me failing to recover because I'd either fall down instead of moving toward the ledge or I'd hit my neutral B instead of my side B with Clairen so it'd do a grab instead of zooming toward the stage. Later in my session when I realized what was happening I tried to make myself take my hands off every direction button and re-input the direction I wanted so it'd work, but the extra thought and delay was enough to mess up recovering still sometimes.
That aside, what's the easiest way to change the SOCD type in HayBox? I am technically on the configurator branch, though I have never used the configurator software. Does that let you change SOCD? Or do I need to edit the source and do a new build?

Soundtoxin, 2024-08-15:
I'm looking in `HayBox/src/modes/RivalsOfAether.cpp` and there's not a single instance of `socd` in here. I looked in `Melee20Button.cpp` which should be the default mode used when I plug my controller in. I see it has `_horizontal_socd = false;` but using [gamepad tester](https://hardwaretester.com/gamepad) I see the same behavior as the Rivals mode where if I hold left, then right, then release right, it goes to neutral instead of back to left.
However I've noticed in Ultimate mode that the SOCD works how I want/expect with gamepad tester.
`Ultimate.cpp` also does not have `socd` written anywhere... Very confusing.

rehtlaw, 2024-08-21:
at least from my experience owning a rectangle a few years ago, "if I'm holding left, then hold right, then let go of right, it does not go back to holding left, it does nothing instead." matches with my experience back then as well.

Soundtoxin, 2024-08-21:
I don't believe this was the case with my previous firmware version (have not tried reflashing to verify yet as I don't know if downgrades are guaranteed safe and I recently soldered in my switches after they kept popping out of the hotswap sockets so I can't access my BOOTSEL button easily anymore if it goes wrong), and as mentioned there is different behavior that matches what I want in the Ultimate Mode, just not in Rivals Mode (anymore?). So either way it's not controller/firmware-wide.

JonnyHaystack, 2024-08-24:
The default has changed, yes. Not necessarily intentionally nor for any particular reason that I can remember. You can configure this stuff via the configurator protocol now without reflashing.

Soundtoxin, 2024-08-24:
Thanks. I wasn't able to get the configurator running on two different machines, unfortunately. Issue open in relevant repo https://github.com/JonnyHaystack/haybox-cli/issues/6
Glad to know I wasn't crazy and that it should be easy to change once I can get haybox-cli working. I'll close this issue here since you confirmed the default changed and that the configurator can change SOCD settings.

closed 2024-08-24 by Soundtoxin