JonnyHaystack/HayBox #83: Rivals of Aether 2 compatibility

Soundtoxin, 2024-10-30:
Using Rivals 1 mode the Mod Y button doesn't let me walk at all, Mod X works for walking but Mod Y just stops me. Would you suggest a new mode or modifying the Rivals 1 mode or adjusting in-game settings (I heard you can do something with stick deadzones) somehow to make the current one work? Honestly I don't use the walk modifiers much so hasn't been a deal-breaker, but I was wondering if the values for Rivals 1 were generally not a great fit for Rivals 2 now. On that note, would any other existing premade mode be a closer fit do you think?

Soundtoxin, 2024-11-04:
I see there's an open PR ( https://github.com/JonnyHaystack/HayBox/pull/81 ) to get the changes merged. I'm debating whether to try the fork or wait. Does the fork still work with the configurator? I'm running non-default SOCD settings that I'd have to probably set up again, ideally with the configurator the same way I did before.

Soundtoxin, 2025-03-17:
I was gonna close this issue after flashing the new 3.0.0 release and testing it, but Mod-X + Start + R (I tried to figure out the new internal button names in case you had changed the binding, but I think it's just counting left to right, and I reasoned that if B is 1 and Z is 3, then 5 must be the row above, I tried both the far left and far right and neither changed to a valid mode) does not seem to put me in Rivals 2 mode from what I can tell. Using Gamepad tester, the analog stick values aren't changing like they were before with the fork. Is the new mode disabled by default or something? I was gonna poke around with the configurator in case that revealed anything, but getting unpackaged stuff to work on Guix System is a nightmare (pip is not really usable, I tried) and haybox-cli is not working for me at all right now, so I'll have to do that on another machine or mess around more later.

JonnyHaystack, 2025-03-17:
@Soundtoxin if you were on configurator branch before and you want to use updated config defaults you'll have to either add a gamemode config to the on-device config yourself via haybox-cli (I don't think remapp.ing supports this yet) or just nuke flash and reflash HayBox 3.0.

Soundtoxin, 2025-03-17:
Nuke flash is a good idea, hadn't tried that. I just used the flash_nuke file that GP2040-CE ships and then re-flashed HayBox-v3.0.0-pico.uf2 afterward. I'm not noticing a difference, though. I had been running the configurator branch, then flashed over it with Austin's fork based on the pre-configurator version (2.4.0 I think), then flashed 3.0.0 over that. I don't know if my config stuff would've survived all that, but I gave the nuke a try anyway.

Soundtoxin, 2025-03-18:
@JonnyHaystack Unsure if this deserves its own issue, but in addition to the shortcut for Rivals 2 not working, holding Start (MB1) on plugin does not seem to work for going into configurator mode. I am testing on a Windows machine now, `.\haybox devices` outputs nothing, and the controller is still working as a controller when I press buttons with Gamepad Tester open. Perhaps these are related. However, C-Down (RT2) *does* work for bootloader mode, which I think confirms I am on the right version, some bindings may just be messed up. I am using the pico uf2 file, and my controller is an Open-Frame1.

JonnyHaystack, 2025-03-18:
It really doesn't sound like you've actually wiped your flash then. Sounds like you've got old config on there.

Soundtoxin, 2025-03-18:
Well, like I said, holding Start to get into the configurator mode is *not* working, so I can't get at a config, as far as I know. Can you instruct me on how to better wipe the flash? The only configuration I had ever done was change SOCD settings, I am completely fine with going back to defaults if I can manage that.

Neither haybox-cli nor remapp.ing can see my controller at all when I hold Start and plug it in.

Soundtoxin, 2025-03-18:
Scratch that, I tried nuking the flash again from the Windows machine and re-flashing 3.0.0 and it actually works as expected now. I get the full 1.0 analog value in Rivals 2 mode. I guess copying over the uf2 file on my main PC with `cp` didn't work right for some reason. Closing the issue now. Thanks.

closed 2025-03-18 by Soundtoxin