JonnyHaystack/HayBox #104: FGC Mode | cannot assign output to RT4

PineappleBun2, 2025-05-19:
In FgcMode I want to assign my dpadUp to inputs.rt4 (what would normally be c-up in smash mode).

I attempted to add it to FgcMode.cpp like so: `outputs.dpadUp = inputs.rt4;`

However, if I do this when I test it in SF6 (X-input, PC version) nothing happens when I press that button. I then noticed that inputs.rt4 is coupled with inputs.lt1 (what would normally be mod-x in smash mode). I can manage to map the button that is tied to rt4 by mapping lt1 instead: `outputs.dpadUp = inputs.lt1;`
In that case I will get a dpadUp output when I press the rt4 button. The issue is then that the lt1 button will also output a dpadUp when pressed, which results in two buttons being mapped to up. This would make the controller unviable for any tournaments that follow the Capcom ruleset which prohibits mapping the same output to multiple buttons.

Here is the complete FgcMode.cpp that gave me the issue, I didn't modify the code anywhere else. 

```
#include "modes/FgcMode.hpp"

FgcMode::FgcMode() : ControllerMode() {}

void FgcMode::UpdateDigitalOutputs(const InputState &inputs, OutputState &outputs) {
    // Directions
    outputs.dpadLeft = inputs.lf3;
    outputs.dpadRight = inputs.lf1;
    outputs.dpadDown = inputs.lf2;
    outputs.dpadUp = inputs.rt4;

    // Menu keys
    outputs.start = inputs.mb1; 
    outputs.select = inputs.rf6;
    outputs.home = inputs.mb1 && inputs.rf6;
    outputs.leftStickClick = inputs.lt2;
    outputs.rightStickClick = inputs.rt2;

    // Right hand bottom row
    outputs.a = inputs.rt1;
    outputs.b = inputs.rf5;
    outputs.triggerRDigital = inputs.rf7;
    outputs.triggerLDigital = inputs.rf8;

    // Right hand top row
    outputs.x = inputs.rf1;
    outputs.y = inputs.rf2; 
    outputs.buttonR = inputs.rf3;
    outputs.buttonL = inputs.rf4;
}

void FgcMode::UpdateAnalogOutputs(const InputState &inputs, OutputState &outputs) {
    outputs.leftStickX = 128;
    outputs.leftStickY = 128;
    outputs.rightStickX = 128;
    outputs.rightStickY = 128;
    outputs.triggerLAnalog = outputs.triggerLDigital ? 255 : 0;
    outputs.triggerRAnalog = outputs.triggerRDigital ? 255 : 0;
}
```

JonnyHaystack, 2025-05-21:
RT4 is mapped to activate LT1 by default, see config_defaults.hpp. Use remapper software e.g. https://remapp.ing or haybox-cli if you want to change that behaviour.

PineappleBun2, 2025-06-08:
Thank you! I managed to get it to work for me now.